Tsinghua-MARS-Lab/futr3d #8: camera-LiDAR-radar Result

Song-Jingyu, 2022-08-19:
Hi,

Thanks for this amazing work. In the project website I think it's mentioned FUTR3D supports camera-LiDAR-radar fusion but I was not able to find any results and checkpoints mentioned in the paper and github. Could you please suggest the reason/thinking behind this? Thank you.

xyaochen, 2022-08-19:
The camera-LiDAR-Radar fusion results are not promising. It slightly degenerates compared to camera-LiDAR fusion.
I think the radar points in nuScenes are too noisy, so it can't contribute to the results when LiDAR points are available.

Song-Jingyu, 2022-08-19:
Thanks for your quick response. That's good to know!

closed 2022-08-19 by Song-Jingyu

deepmeng, 2023-01-09:
Do you have a model config for 'camera-lidar-radar'?

deepmeng, 2023-01-16:
> Do you have a model config for 'camera-lidar-radar'?

Solved already