Tsinghua-MARS-Lab/futr3d #10: Problem about multi GPUs processing

xinpuliu, 2022-08-27:
在单卡训练的时候可以跑通，但是多卡（4张3090）训练的时候，出现这个问题，能帮我解答下吗？谢谢

Traceback (most recent call last):
  File "tools/train.py", line 264, in <module>
    main()
  File "tools/train.py", line 260, in main
    meta=meta)
  File "/home/fuwuqianyan/LXP/mmdetection3d/mmdet3d/apis/train.py", line 351, in train_model
    meta=meta)
  File "/home/fuwuqianyan/LXP/mmdetection3d/mmdet3d/apis/train.py", line 319, in train_detector
    runner.run(data_loaders, cfg.workflow)
  File "/home/fuwuqianyan/anaconda3/envs/wunai3/lib/python3.7/site-packages/mmcv/runner/epoch_based_runner.py", line 136, in run
    epoch_runner(data_loaders[i], **kwargs)
  File "/home/fuwuqianyan/anaconda3/envs/wunai3/lib/python3.7/site-packages/mmcv/runner/epoch_based_runner.py", line 49, in train
    for i, data_batch in enumerate(self.data_loader):
  File "/home/fuwuqianyan/anaconda3/envs/wunai3/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 355, in __iter__
    return self._get_iterator()
  File "/home/fuwuqianyan/anaconda3/envs/wunai3/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 301, in _get_iterator
    return _MultiProcessingDataLoaderIter(self)
  File "/home/fuwuqianyan/anaconda3/envs/wunai3/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 914, in __init__
    w.start()
  File "/home/fuwuqianyan/anaconda3/envs/wunai3/lib/python3.7/multiprocessing/process.py", line 112, in start
    self._popen = self._Popen(self)
  File "/home/fuwuqianyan/anaconda3/envs/wunai3/lib/python3.7/multiprocessing/context.py", line 223, in _Popen
    return _default_context.get_context().Process._Popen(process_obj)
  File "/home/fuwuqianyan/anaconda3/envs/wunai3/lib/python3.7/multiprocessing/context.py", line 284, in _Popen
    return Popen(process_obj)
  File "/home/fuwuqianyan/anaconda3/envs/wunai3/lib/python3.7/multiprocessing/popen_spawn_posix.py", line 32, in __init__
    super().__init__(process_obj)
  File "/home/fuwuqianyan/anaconda3/envs/wunai3/lib/python3.7/multiprocessing/popen_fork.py", line 20, in __init__
    self._launch(process_obj)
  File "/home/fuwuqianyan/anaconda3/envs/wunai3/lib/python3.7/multiprocessing/popen_spawn_posix.py", line 47, in _launch
    reduction.dump(process_obj, fp)
  File "/home/fuwuqianyan/anaconda3/envs/wunai3/lib/python3.7/multiprocessing/reduction.py", line 61, in dump
    ForkingPickler(file, protocol).dump(obj)
TypeError: can't pickle dict_keys objects
Traceback (most recent call last):
  File "tools/train.py", line 264, in <module>
    main()
  File "tools/train.py", line 260, in main
    meta=meta)
  File "/home/fuwuqianyan/LXP/mmdetection3d/mmdet3d/apis/train.py", line 351, in train_model
    meta=meta)
  File "/home/fuwuqianyan/LXP/mmdetection3d/mmdet3d/apis/train.py", line 319, in train_detector
    runner.run(data_loaders, cfg.workflow)
  File "/home/fuwuqianyan/anaconda3/envs/wunai3/lib/python3.7/site-packages/mmcv/runner/epoch_based_runner.py", line 136, in run
    epoch_runner(data_loaders[i], **kwargs)
  File "/home/fuwuqianyan/anaconda3/envs/wunai3/lib/python3.7/site-packages/mmcv/runner/epoch_based_runner.py", line 49, in train
    for i, data_batch in enumerate(self.data_loader):
  File "/home/fuwuqianyan/anaconda3/envs/wunai3/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 355, in __iter__
    return self._get_iterator()
  File "/home/fuwuqianyan/anaconda3/envs/wunai3/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 301, in _get_iterator
    return _MultiProcessingDataLoaderIter(self)
  File "/home/fuwuqianyan/anaconda3/envs/wunai3/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 914, in __init__
    w.start()
  File "/home/fuwuqianyan/anaconda3/envs/wunai3/lib/python3.7/multiprocessing/process.py", line 112, in start
    self._popen = self._Popen(self)
  File "/home/fuwuqianyan/anaconda3/envs/wunai3/lib/python3.7/multiprocessing/context.py", line 223, in _Popen
    return _default_context.get_context().Process._Popen(process_obj)
  File "/home/fuwuqianyan/anaconda3/envs/wunai3/lib/python3.7/multiprocessing/context.py", line 284, in _Popen
    return Popen(process_obj)
  File "/home/fuwuqianyan/anaconda3/envs/wunai3/lib/python3.7/multiprocessing/popen_spawn_posix.py", line 32, in __init__
    super().__init__(process_obj)
  File "/home/fuwuqianyan/anaconda3/envs/wunai3/lib/python3.7/multiprocessing/popen_fork.py", line 20, in __init__
    self._launch(process_obj)
  File "/home/fuwuqianyan/anaconda3/envs/wunai3/lib/python3.7/multiprocessing/popen_spawn_posix.py", line 47, in _launch
    reduction.dump(process_obj, fp)
  File "/home/fuwuqianyan/anaconda3/envs/wunai3/lib/python3.7/multiprocessing/reduction.py", line 61, in dump
    ForkingPickler(file, protocol).dump(obj)
TypeError: can't pickle dict_keys objects
Killing subprocess 59448
Killing subprocess 59449
Killing subprocess 59450
Killing subprocess 59451
Traceback (most recent call last):
  File "/home/fuwuqianyan/anaconda3/envs/wunai3/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/home/fuwuqianyan/anaconda3/envs/wunai3/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/fuwuqianyan/anaconda3/envs/wunai3/lib/python3.7/site-packages/torch/distributed/launch.py", line 340, in <module>
    main()
  File "/home/fuwuqianyan/anaconda3/envs/wunai3/lib/python3.7/site-packages/torch/distributed/launch.py", line 326, in main
    sigkill_handler(signal.SIGTERM, None)  # not coming back
  File "/home/fuwuqianyan/anaconda3/envs/wunai3/lib/python3.7/site-packages/torch/distributed/launch.py", line 301, in sigkill_handler
    raise subprocess.CalledProcessError(returncode=last_return_code, cmd=cmd)
subprocess.CalledProcessError: Command '['/home/fuwuqianyan/anaconda3/envs/wunai3/bin/python3', '-u', 'tools/train.py', '--local_rank=3', 'plugin/futr3d/configs/lidar_only/01voxel_q6_step_38e.py', '--launcher', 'pytorch']' returned non-zero exit status 1.

xinpuliu, 2022-08-27:
我的环境如下

sys.platform: linux
Python: 3.7.13 (default, Mar 29 2022, 02:18:16) [GCC 7.5.0]
CUDA available: True
GPU 0,1,2,3: NVIDIA GeForce RTX 3090
CUDA_HOME: /usr/local/cuda-11.3
NVCC: Cuda compilation tools, release 11.3, V11.3.58
GCC: gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
PyTorch: 1.8.1+cu111
PyTorch compiling details: PyTorch built with:
  - GCC 7.3
  - C++ Version: 201402
  - Intel(R) Math Kernel Library Version 2020.0.0 Product Build 20191122 for Intel(R) 64 architecture applications
  - Intel(R) MKL-DNN v1.7.0 (Git Hash 7aed236906b1f7a05c0917e5257a1af05e9ff683)
  - OpenMP 201511 (a.k.a. OpenMP 4.5)
  - NNPACK is enabled
  - CPU capability usage: AVX2
  - CUDA Runtime 11.1
  - NVCC architecture flags: -gencode;arch=compute_37,code=sm_37;-gencode;arch=compute_50,code=sm_50;-gencode;arch=compute_60,code=sm_60;-gencode;arch=compute_70,code=sm_70;-gencode;arch=compute_75,code=sm_75;-gencode;arch=compute_80,code=sm_80;-gencode;arch=compute_86,code=sm_86
  - CuDNN 8.0.5
  - Magma 2.5.2

TorchVision: 0.9.1+cu111
OpenCV: 4.6.0
MMCV: 1.6.1
MMCV Compiler: GCC 7.3
MMCV CUDA Compiler: 11.1
MMDetection: 2.25.1
MMSegmentation: 0.27.0
MMDetection3D: 1.0.0rc4+1afff87
spconv2.0: True

xyaochen, 2022-08-27:
我之前也遇到过这个问题，这似乎是mmdet库中的一个Bug。
有三种解决方案：
1. 在config里令workers_per_gpu=0
2. 修改mmcv中runner的代码，将https://github.com/open-mmlab/mmcv/blob/e7f049afa12bb7607d44b15dd3d7daa4e6394b0a/mmcv/runner/dist_utils.py#L16 中的spawn改成fork
3. 在tools/train.py  (https://github.com/Tsinghua-MARS-Lab/futr3d/blob/main/tools/train.py#L262) 中加 torch.multiprocessing.set_start_method('fork')，比如：
`if __name__ == '__main__':  
    torch.multiprocessing.set_start_method('fork')
    main()
`

xinpuliu, 2022-08-27:
> 我之前也遇到过这个问题，这似乎是mmdet库中的一个Bug。 有三种解决方案：
>
> 1. 在config里令workers_per_gpu=0
> 2. [...]

非常感谢，根据您提供的方法，bug已经解决，可以多卡并行跑起来了。
另外还有个小问题想问一下，您可否提供下FUTR3D所用的Nuscene数据集的文件夹结构呀，我不知道这个结构是怎样的，我也是第一次跑Nuscene数据集，结构是按照网上来的，不知道对不对。

xyaochen, 2022-08-28:
> 非常感谢，根据您提供的方法，bug已经解决，可以多卡并行跑起来了。 另外还有个小问题想问一下，您可否提供下FUTR3D所用的Nuscene数据集的文件夹结构呀，我不知道这个结构是怎样的，我也是第一次跑Nuscene数据集，结构是按照网上来的，不知道对不对。

文件夹结构就是按照mmdet3d官方的结构，没有变化

xinpuliu, 2022-08-28:
好的明白  感谢！

closed 2022-08-28 by xinpuliu