Tsinghua-MARS-Lab/futr3d #11: KeyError: 'FUTR3D is not in the detector registry'

shb9793, 2022-08-31:
When I try to fine-tune  `futr3d` config to KITTI dataset, the error occurs as follows: 

```
2022-08-31 15:46:14,605 - mmdet - INFO - Set random seed to 0, deterministic: False
Traceback (most recent call last):
  File "tools/train.py", line 236, in <module>
    main()
  File "tools/train.py", line 200, in main
    test_cfg=cfg.get('test_cfg'))
  File "/data/run01/scz3687/openmmlab0130/mmdetection3d/mmdet3d/models/builder.py", line 52, in build_detector
    return build(cfg, DETECTORS, dict(train_cfg=train_cfg, test_cfg=test_cfg))
  File "/HOME/scz3687/.conda/envs/open-mmlab0130/lib/python3.7/site-packages/mmdet/models/builder.py", line 34, in build
    return build_from_cfg(cfg, registry, default_args)
  File "/HOME/scz3687/.conda/envs/open-mmlab0130/lib/python3.7/site-packages/mmcv/utils/registry.py", line 172, in build_from_cfg
    f'{obj_type} is not in the {registry.name} registry')
KeyError: 'FUTR3D is not in the detector registry'
```

I have try to recompile the mmdet3d framework, but I it didn't work.

shb9793, 2022-08-31:
These are some details about my environment:

```
plugin
2022-08-31 15:46:09,559 - mmdet - INFO - Environment info:
------------------------------------------------------------
sys.platform: linux
Python: 3.7.13 (default, Mar 29 2022, 02:18:16) [GCC 7.5.0]
CUDA available: True
GPU 0: NVIDIA GeForce RTX 3090
CUDA_HOME: /data/apps/cuda/11.1
NVCC: Build cuda_11.1.TC455_06.29069683_0
GCC: gcc (GCC) 4.8.5 20150623 (Red Hat 4.8.5-44)
PyTorch: 1.8.0
PyTorch compiling details: PyTorch built with:
  - GCC 7.3
  - C++ Version: 201402
  - Intel(R) oneAPI Math Kernel Library Version 2021.4-Product Build 20210904 for Intel(R) 64 architecture applications
  - Intel(R) MKL-DNN v1.7.0 (Git Hash 7aed236906b1f7a05c0917e5257a1af05e9ff683)
  - OpenMP 201511 (a.k.a. OpenMP 4.5)
  - NNPACK is enabled
  - CPU capability usage: AVX2
  - CUDA Runtime 11.1
  - NVCC architecture flags: -gencode;arch=compute_37,code=sm_37;-gencode;arch=compute_50,code=sm_50;-gencode;arch=compute_60,code=sm_60;-gencode;arch=compute_61,code=sm_61;-gencode;arch=compute_70,code=sm_70;-gencode;arch=compute_75,code=sm_75;-gencode;arch=compute_80,code=sm_80;-gencode;arch=compute_86,code=sm_86;-gencode;arch=compute_37,code=compute_37
  - CuDNN 8.0.5
  - Magma 2.5.2
  - Build settings: BLAS_INFO=mkl, BUILD_TYPE=Release, CUDA_VERSION=11.1, CUDNN_VERSION=8.0.5, CXX_COMPILER=/opt/rh/devtoolset-7/root/usr/bin/c++, CXX_FLAGS= -Wno-deprecated -fvisibility-inlines-hidden -DUSE_PTHREADPOOL -fopenmp -DNDEBUG -DUSE_KINETO -DUSE_FBGEMM -DUSE_QNNPACK -DUSE_PYTORCH_QNNPACK -DUSE_XNNPACK -O2 -fPIC -Wno-narrowing -Wall -Wextra -Werror=return-type -Wno-missing-field-initializers -Wno-type-limits -Wno-array-bounds -Wno-unknown-pragmas -Wno-sign-compare -Wno-unused-parameter -Wno-unused-variable -Wno-unused-function -Wno-unused-result -Wno-unused-local-typedefs -Wno-strict-overflow -Wno-strict-aliasing -Wno-error=deprecated-declarations -Wno-stringop-overflow -Wno-psabi -Wno-error=pedantic -Wno-error=redundant-decls -Wno-error=old-style-cast -fdiagnostics-color=always -faligned-new -Wno-unused-but-set-variable -Wno-maybe-uninitialized -fno-math-errno -fno-trapping-math -Werror=format -Wno-stringop-overflow, LAPACK_INFO=mkl, PERF_WITH_AVX=1, PERF_WITH_AVX2=1, PERF_WITH_AVX512=1, TORCH_VERSION=1.8.0, USE_CUDA=ON, USE_CUDNN=ON, USE_EXCEPTION_PTR=1, USE_GFLAGS=OFF, USE_GLOG=OFF, USE_MKL=ON, USE_MKLDNN=ON, USE_MPI=OFF, USE_NCCL=ON, USE_NNPACK=ON, USE_OPENMP=ON, 

TorchVision: 0.9.0
OpenCV: 4.6.0
MMCV: 1.2.6
MMCV Compiler: GCC 5.4
MMCV CUDA Compiler: not available
MMDetection: 2.10.0
MMDetection3D: 0.13.0+b255d16
```

shb9793, 2022-08-31:
And this is the command of implementation: 

```
#!/bin/bash
#SBATCH --gpus=1

module load cuda/11.1
module load anaconda/2021.05

cd /HOME/scz3687/run/openmmlab0130/mmdetection3d

export PYTHONUNBUFFERED=1
source activate open-mmlab0130

python tools/train.py plugin/futr3d/configs/lidar_cam/res101_01voxel_step_3e_kitti.py --work-dir /HOME/scz3687/run/openmmlab0130/mmdetection3d/work_dir/futr3d_kitti
```

xyaochen, 2022-08-31:
It seems like you didn't import the plugin module correctly. You can check this code (https://github.com/Tsinghua-MARS-Lab/futr3d/blob/main/tools/train.py#L107).

closed 2022-09-05 by shb9793

cgl-cell, 2024-02-19:
I had the same problem, have you solved it yet?