Tsinghua-MARS-Lab/futr3d #12: Confused about "return  " at plugin/futr3d/models/backbones/radar_encoder.py#L93

LeopoldACC, 2022-09-02:
Hi， @xyaochen ，thanks for your work in multi sensor fusion detection.
But I'm confused about the implementation in [plugin/futr3d/models/backbones/radar_encoder.py#L93](https://github.com/Tsinghua-MARS-Lab/futr3d/blob/5c290b699ddc5098141414921da6d0d3fbad4b5e/plugin/futr3d/models/backbones/radar_encoder.py#L93)
If implement like that, the module registried as "radar_encoder" will return nothing.Can you help me understand it?Thanks a lot!
```python
    def forward(self, points):
        '''
        points: [B, N, C]. N: as max
        masks: [B, N, 1]
        ret: 
            out: [B, N, C+1], last channel as 0-1 mask
        '''
        masks = points[:, :, [-1]]
        x = points[:, :, :-1]
        xy = points[:, :, :2]
        
        for feat_layer in self.feat_layers:
            x = feat_layer(x)
        
        out = x * masks

        out = torch.cat((x, masks), dim=-1)

        out = torch.cat((xy, out), dim=-1)
        return 
```

xyaochen, 2022-09-05:
Sorry for the typo! It should be "return out"

LeopoldACC, 2022-09-05:
# code's change
```bash
diff --git a/plugin/futr3d/configs/cam_radar/res101_radar.py b/plugin/futr3d/configs/cam_radar/res101_radar.py
index 79bfd48..6adab02 100644
--- a/plugin/futr3d/configs/cam_radar/res101_radar.py
+++ b/plugin/futr3d/configs/cam_radar/res101_radar.py
@@ -1,6 +1,10 @@
+# _base_ = [^M
+#     '../../../../configs/_base_/datasets/nus-3d.py',
+#     '../../../../configs/_base_/default_runtime.py'
+# ]^M
 _base_ = [
-    '../../../../configs/_base_/datasets/nus-3d.py',
-    '../../../../configs/_base_/default_runtime.py'
+    '../../../../plugin/_base_/datasets/nus-3d.py',
+    '../../../../plugin/_base_/default_runtime.py'
 ]
 plugin=True
 plugin_dir='plugin/futr3d/'
@@ -288,7 +292,7 @@ data = dict(
         # dataset=dict(
             type=dataset_type,
             data_root=data_root,
-            ann_file=data_root + 'radar_nuscenes_5sweeps_infos_train_radar.pkl',
+            ann_file=data_root + 'radar_nuscenes_5sweeps_infos_train.pkl',
             pipeline=train_pipeline,
             classes=class_names,
             modality=input_modality,
@@ -301,11 +305,11 @@ data = dict(
     val=dict(
             type=dataset_type,
             pipeline=test_pipeline, classes=class_names, modality=input_modality,
-            ann_file=data_root + 'radar_nuscenes_5sweeps_infos_val_radar.pkl',),
+            ann_file=data_root + 'radar_nuscenes_5sweeps_infos_val.pkl',),^M
     test=dict(
             type=dataset_type,
             pipeline=test_pipeline, classes=class_names, modality=input_modality,
-            ann_file=data_root + 'radar_nuscenes_5sweeps_infos_val_radar.pkl',),)
+            ann_file=data_root + 'radar_nuscenes_5sweeps_infos_val.pkl',),)^M
 
 optimizer = dict(
     type='AdamW',
diff --git a/plugin/futr3d/models/backbones/radar_encoder.py b/plugin/futr3d/models/backbones/radar_encoder.py
index 9866afb..fc10d5b 100644
--- a/plugin/futr3d/models/backbones/radar_encoder.py
+++ b/plugin/futr3d/models/backbones/radar_encoder.py
@@ -78,6 +78,8 @@ class RadarPointEncoder(nn.Module):
         ret: 
             out: [B, N, C+1], last channel as 0-1 mask
         '''
+        if isinstance(points,list):^M
+            points = points[0]^M
         masks = points[:, :, [-1]]
         x = points[:, :, :-1]
         xy = points[:, :, :2]
@@ -90,4 +92,4 @@ class RadarPointEncoder(nn.Module):
         out = torch.cat((x, masks), dim=-1)
 
         out = torch.cat((xy, out), dim=-1)
-        return 
\ No newline at end of file
+        return out
\ No newline at end of file
diff --git a/plugin/futr3d/models/utils/attention.py b/plugin/futr3d/models/utils/attention.py
index a945a90..af59e08 100644
--- a/plugin/futr3d/models/utils/attention.py
+++ b/plugin/futr3d/models/utils/attention.py
@@ -267,6 +267,7 @@ class FUTR3DCrossAtten(BaseModule):
             pts_output = self.pts_output_proj(pts_output)
         
[... 61 lines cut ...]
bicycle 0.000   1.000   1.000   1.000   1.000   1.000
traffic_cone    0.000   1.000   1.000   nan     nan     nan
barrier 0.000   1.000   1.000   1.000   nan     nan
{'pts_bbox_NuScenes/car_AP_dist_0.5': 0.0, 'pts_bbox_NuScenes/car_AP_dist_1.0': 0.0, 'pts_bbox_NuScenes/car_AP_dist_2.0': 0.0, 'pts_bbox_NuScenes/car_AP_dist_4.0': 0.0, 'pts_bbox_NuScenes/car_trans_err':
1.0, 'pts_bbox_NuScenes/car_scale_err': 1.0, 'pts_bbox_NuScenes/car_orient_err': 1.0, 'pts_bbox_NuScenes/car_vel_err': 1.0, 'pts_bbox_NuScenes/car_attr_err': 1.0, 'pts_bbox_NuScenes/truck_AP_dist_0.5': 0.
0, 'pts_bbox_NuScenes/truck_AP_dist_1.0': 0.0, 'pts_bbox_NuScenes/truck_AP_dist_2.0': 0.0, 'pts_bbox_NuScenes/truck_AP_dist_4.0': 0.0, 'pts_bbox_NuScenes/truck_trans_err': 1.0, 'pts_bbox_NuScenes/truck_sc
ale_err': 1.0, 'pts_bbox_NuScenes/truck_orient_err': 1.0, 'pts_bbox_NuScenes/truck_vel_err': 1.0, 'pts_bbox_NuScenes/truck_attr_err': 1.0, 'pts_bbox_NuScenes/construction_vehicle_AP_dist_0.5': 0.0, 'pts_b
box_NuScenes/construction_vehicle_AP_dist_1.0': 0.0, 'pts_bbox_NuScenes/construction_vehicle_AP_dist_2.0': 0.0, 'pts_bbox_NuScenes/construction_vehicle_AP_dist_4.0': 0.0, 'pts_bbox_NuScenes/construction_v
ehicle_trans_err': 1.0, 'pts_bbox_NuScenes/construction_vehicle_scale_err': 1.0, 'pts_bbox_NuScenes/construction_vehicle_orient_err': 1.0, 'pts_bbox_NuScenes/construction_vehicle_vel_err': 1.0, 'pts_bbox_
NuScenes/construction_vehicle_attr_err': 1.0, 'pts_bbox_NuScenes/bus_AP_dist_0.5': 0.0, 'pts_bbox_NuScenes/bus_AP_dist_1.0': 0.0, 'pts_bbox_NuScenes/bus_AP_dist_2.0': 0.0, 'pts_bbox_NuScenes/bus_AP_dist_4
.0': 0.0, 'pts_bbox_NuScenes/bus_trans_err': 1.0, 'pts_bbox_NuScenes/bus_scale_err': 1.0, 'pts_bbox_NuScenes/bus_orient_err': 1.0, 'pts_bbox_NuScenes/bus_vel_err': 1.0, 'pts_bbox_NuScenes/bus_attr_err': 1
.0, 'pts_bbox_NuScenes/trailer_AP_dist_0.5': 0.0, 'pts_bbox_NuScenes/trailer_AP_dist_1.0': 0.0, 'pts_bbox_NuScenes/trailer_AP_dist_2.0': 0.0, 'pts_bbox_NuScenes/trailer_AP_dist_4.0': 0.0, 'pts_bbox_NuScen
es/trailer_trans_err': 1.0, 'pts_bbox_NuScenes/trailer_scale_err': 1.0, 'pts_bbox_NuScenes/trailer_orient_err': 1.0, 'pts_bbox_NuScenes/trailer_vel_err': 1.0, 'pts_bbox_NuScenes/trailer_attr_err': 1.0, 'p
ts_bbox_NuScenes/barrier_AP_dist_0.5': 0.0, 'pts_bbox_NuScenes/barrier_AP_dist_1.0': 0.0, 'pts_bbox_NuScenes/barrier_AP_dist_2.0': 0.0, 'pts_bbox_NuScenes/barrier_AP_dist_4.0': 0.0, 'pts_bbox_NuScenes/bar
[... cut ...]
box_NuScenes/pedestrian_attr_err': 0.7047, 'pts_bbox_NuScenes/traffic_cone_AP_dist_0.5': 0.0, 'pts_bbox_NuScenes/traffic_cone_AP_dist_1.0': 0.0, 'pts_bbox_NuScenes/traffic_cone_AP_dist_2.0': 0.0, 'pts_bbo
x_NuScenes/traffic_cone_AP_dist_4.0': 0.0, 'pts_bbox_NuScenes/traffic_cone_trans_err': 1.0, 'pts_bbox_NuScenes/traffic_cone_scale_err': 1.0, 'pts_bbox_NuScenes/traffic_cone_orient_err': nan, 'pts_bbox_NuS
cenes/traffic_cone_vel_err': nan, 'pts_bbox_NuScenes/traffic_cone_attr_err': nan, 'pts_bbox_NuScenes/NDS': 0.009765414502821469, 'pts_bbox_NuScenes/mAP': 0.0}
```

Song-Jingyu, 2022-09-20:
Want to follow up here as the code looks a little weird here to me,

```
        out = x * masks

        out = torch.cat((x, masks), dim=-1)

        out = torch.cat((xy, out), dim=-1)
        return 
```

What is the motivation of out=x*masks here since the out got overwrite later on. Thanks!

closed 2023-05-29 by xyaochen