Tsinghua-MARS-Lab/futr3d #13: ImportError: cannot import name 'build' from 'mmcv.utils.registry'

shb9793, 2022-09-05:
The detail about is showed as follows:

```ruby
Traceback (most recent call last):
  File "tools/train.py", line 17, in <module>
    from mmdet3d.models.builder import build_detector
  File "/data/run01/scz3687/openmmlab0130/mmdetection3d/mmdet3d/models/__init__.py", line 1, in <module>
    from .backbones import *  # noqa: F401,F403
  File "/data/run01/scz3687/openmmlab0130/mmdetection3d/mmdet3d/models/backbones/__init__.py", line 3, in <module>
    from .nostem_regnet import NoStemRegNet
  File "/data/run01/scz3687/openmmlab0130/mmdetection3d/mmdet3d/models/backbones/nostem_regnet.py", line 2, in <module>
    from ..builder import BACKBONES
  File "/data/run01/scz3687/openmmlab0130/mmdetection3d/mmdet3d/models/builder.py", line 3, in <module>
    from mmcv.utils.registry import build
ImportError: cannot import name 'build' from 'mmcv.utils.registry' (/HOME/scz3687/.conda/envs/open-mmlab0130/lib/python3.7/site-packages/mmcv/utils/registry.py)
```

My environment:
mmcv-full==1.3.14, mmdet==2.14.0, mmdet3d==0.13.0

shb9793, 2022-09-05:
Could you please tell me the exact version about your environment? I have try different version of `mmcv-full` and  `mmdet`. But all of them didn't work as the aforementioned.

xyaochen, 2022-09-05:
Have you rewritten the mmdet3d code? The source code seems different from your bug report
https://github.com/open-mmlab/mmdetection3d/blob/v0.13.0/mmdet3d/models/builder.py#L2
There is no "from mmcv.utils.registry import build" in line 3

shb9793, 2022-09-05:
Yeap~Because I hava faced the problem as follows, if I use original code as https://github.com/open-mmlab/mmdetection3d/blob/v0.13.0/mmdet3d/models/builder.py#L2

```ruby
Traceback (most recent call last):
  File "tools/train.py", line 17, in <module>
    from mmdet3d.models import build_detector
  File "/data/run01/scz3687/openmmlab0130/mmdetection3d/mmdet3d/models/__init__.py", line 1, in <module>
    from .backbones import *  # noqa: F401,F403
  File "/data/run01/scz3687/openmmlab0130/mmdetection3d/mmdet3d/models/backbones/__init__.py", line 3, in <module>
    from .nostem_regnet import NoStemRegNet
  File "/data/run01/scz3687/openmmlab0130/mmdetection3d/mmdet3d/models/backbones/nostem_regnet.py", line 2, in <module>
    from ..builder import BACKBONES
  File "/data/run01/scz3687/openmmlab0130/mmdetection3d/mmdet3d/models/builder.py", line 4, in <module>
    from mmdet.models.builder import (BACKBONES, DETECTORS, HEADS, LOSSES, NECKS,
ImportError: cannot import name 'build' from 'mmdet.models.builder' (/HOME/scz3687/.conda/envs/open-mmlab0130/lib/python3.7/site-packages/mmdet/models/builder.py)
```

So I tried to import `build` module from `mmcv.utils.registry` part.

xyaochen, 2022-09-05:
https://mmdetection3d.readthedocs.io/en/stable/getting_started.html
You can check this. mmdet3d==0.13.0 is not compatible with mmdet==2.14.0. You can try to change the version as above

shb9793, 2022-09-05:
Thanks a lot for your reply.  I have tried mmdet==2.11.0, but it didn't work well. Because `inverse_sigmoid` module in `mmdet/models/utils/transformer.py` didn't exist.

```ruby
plugin.futr3d
Traceback (most recent call last):
  File "tools/train.py", line 236, in <module>
    main()
  File "tools/train.py", line 119, in main
    plg_lib = importlib.import_module(_module_path)
  File "/HOME/scz3687/.conda/envs/open-mmlab0130/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/data/run01/scz3687/openmmlab0130/mmdetection3d/plugin/futr3d/__init__.py", line 5, in <module>
    from .models.dense_head.detr_mdfs_head import DeformableFUTR3DHead
  File "/data/run01/scz3687/openmmlab0130/mmdetection3d/plugin/futr3d/models/dense_head/detr_mdfs_head.py", line 12, in <module>
    from mmdet.models.utils.transformer import inverse_sigmoid
ImportError: cannot import name 'inverse_sigmoid' from 'mmdet.models.utils.transformer' (/HOME/scz3687/.conda/envs/open-mmlab0130/lib/python3.7/site-packages/mmdet/models/utils/transformer.py)
```

xyaochen, 2022-09-05:
You can import inver_sigmoid from here "https://github.com/Tsinghua-MARS-Lab/futr3d/blob/main/plugin/futr3d/models/utils/attention.py#L24"

shb9793, 2022-09-05:
Thanks a lot! Maybe this is the exact solution for this problem. Best wishes!

closed 2023-05-29 by xyaochen