Tsinghua-MARS-Lab/futr3d #14: KeyError: "FUTR3D: 'FPNV2 is not in the models registry'"

drilistbox, 2022-09-05:
Hi:
    I encounter the error in title before. It seems that the FPNV2 has not been defined in the relese paper. Will you update this? 
                                                                                                                                                                                                  Thanks a lot!

Song-Jingyu, 2022-09-17:
The same problem here! I ended up using FPN instead of FPNV2. But I am not certain if this is the right practice. Hope the authors could clarify.

xyaochen, 2022-09-19:
I have added FPNV2.
The only difference between FPNV2 and original FPN is https://github.com/Tsinghua-MARS-Lab/futr3d/blob/main/plugin/futr3d/models/necks/fpn.py#L157. This is aimed to solve https://github.com/Tsinghua-MARS-Lab/futr3d/issues/16 and will not affect the forward results.

shb9793, 2022-09-20:
> I have added FPNV2. The only difference between FPNV2 and original FPN is https://github.com/Tsinghua-MARS-Lab/futr3d/blob/main/plugin/futr3d/models/necks/fpn.py#L157. This is aimed to solve #16 and will not affect the forward results.

Thanks a lot for your contribution! I have faced this problem yesterday.

closed 2023-05-29 by xyaochen