Tsinghua-MARS-Lab/futr3d #17: ImportError: cannot import name 'DeformableDETRHead' from 'mmdet.models.dense_heads'

shb9793, 2022-09-16:
完整报错如下：

```ruby
plugin.futr3d
Traceback (most recent call last):
  File "tools/train.py", line 236, in <module>
    main()
  File "tools/train.py", line 119, in main
    plg_lib = importlib.import_module(_module_path)
  File "/HOME/scz3687/.conda/envs/open-mmlab0130/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/data/run01/scz3687/openmmlab0130/mmdetection3d/plugin/futr3d/__init__.py", line 5, in <module>
    from .models.dense_head.detr_mdfs_head import DeformableFUTR3DHead
  File "/data/run01/scz3687/openmmlab0130/mmdetection3d/plugin/futr3d/models/dense_head/detr_mdfs_head.py", line 15, in <module>
    from mmdet.models.dense_heads import DeformableDETRHead, DETRHead
ImportError: cannot import name 'DeformableDETRHead' from 'mmdet.models.dense_heads' (/HOME/scz3687/.conda/envs/open-mmlab0130/lib/python3.7/site-packages/mmdet/models/dense_heads/__init__.py)
```

我去官方的仓库看了一下，mmdet==2.11.0里刚好没有这两个包，但是mmdet==2.12.0以后的版本就有了

可是mmdet3d==0.13.0的安装要求是mmdet>=2.10.0, <=2.11.0

请问这个问题该怎么解决呢

closed 2022-09-28 by shb9793

zen-star, 2022-10-08:
@shb9793 请问你是如何解决的呢？

shb9793, 2022-10-09:
> @shb9793 请问你是如何解决的呢？

我换了个mmdet==2.14.0的版本，我想不到其他好办法了