Tsinghua-MARS-Lab/futr3d #18: RuntimeError: shape '[1, 4, 1, 4, 4]' is invalid for input of size 16

shb9793, 2022-09-29:
I want to train this model on my customized dataset (a camera + a LIDAR), and I have transform it to  'KITTI' format. 
But I don't know the exact source and solution of the following error.

```ruby
2022-09-29 17:52:53,045 - mmdet - INFO - workflow: [('train', 1)], max: 3 epochs
/HOME/scz3687/.conda/envs/open-mmlab0171/lib/python3.7/site-packages/torch/nn/functional.py:718: UserWarning: Named tensors and all their associated APIs are an experimental feature and subject to change. Please do not use them for anything important until they are released as stable. (Triggered internally at  /pytorch/c10/core/TensorImpl.h:1156.)
  return torch.max_pool2d(input, kernel_size, stride, padding, dilation, ceil_mode)
Traceback (most recent call last):
  File "tools/train.py", line 248, in <module>
    main()
  File "tools/train.py", line 244, in main
    meta=meta)
  File "/data/run01/scz3687/openmmlab0171/mmdetection3d/mmdet3d/apis/train.py", line 35, in train_model
    meta=meta)
  File "/HOME/scz3687/.conda/envs/open-mmlab0171/lib/python3.7/site-packages/mmdet/apis/train.py", line 170, in train_detector
    runner.run(data_loaders, cfg.workflow)
  File "/HOME/scz3687/.conda/envs/open-mmlab0171/lib/python3.7/site-packages/mmcv/runner/epoch_based_runner.py", line 127, in run
    epoch_runner(data_loaders[i], **kwargs)
  File "/HOME/scz3687/.conda/envs/open-mmlab0171/lib/python3.7/site-packages/mmcv/runner/epoch_based_runner.py", line 50, in train
    self.run_iter(data_batch, train_mode=True, **kwargs)
  File "/HOME/scz3687/.conda/envs/open-mmlab0171/lib/python3.7/site-packages/mmcv/runner/epoch_based_runner.py", line 30, in run_iter
    **kwargs)
  File "/HOME/scz3687/.conda/envs/open-mmlab0171/lib/python3.7/site-packages/mmcv/parallel/data_parallel.py", line 67, in train_step
    return self.module.train_step(*inputs[0], **kwargs[0])
  File "/HOME/scz3687/.conda/envs/open-mmlab0171/lib/python3.7/site-packages/mmdet/models/detectors/base.py", line 237, in train_step
    losses = self(**data)
  File "/HOME/scz3687/.conda/envs/open-mmlab0171/lib/python3.7/site-packages/torch/nn/modules/module.py", line 1051, in _call_impl
    return forward_call(*input, **kwargs)
  File "/HOME/scz3687/.conda/envs/open-mmlab0171/lib/python3.7/site-packages/mmcv/runner/fp16_utils.py", line 98, in new_func
    return old_func(*args, **kwargs)
  File "/data/run01/scz3687/openmmlab0171/mmdetection3d/mmdet3d/models/detectors/base.py", line 59, in forward
    return self.forward_train(**kwargs)
  File "/data/run01/scz3687/openmmlab0171/mmdetection3d/plugin/futr3d/models/detectors/futr3d.py", line 200, in forward_train
    gt_bboxes_ignore)
  File "/data/run01/scz3687/openmmlab0171/mmdetection3d/plugin/futr3d/models/detectors/futr3d.py", line 135, in forward_mdfs_train
    outs = self.pts_bbox_head(pts_feats, img_feats, rad_feats, img_metas)
  File "/HOME/scz3687/.conda/envs/open-mmlab0171/lib/python3.7/site-packages/torch/nn/modules/module.py", line 1051, in _call_impl
    return forward_call(*input, **kwargs)
  File "/data/run01/scz3687/openmmlab0171/mmdetection3d/plugin/futr3d/models/dense_head/detr_mdfs_head.py", line 130, in forward
    img_metas=img_metas,
  File "/HOME/scz3687/.conda/envs/open-mmlab0171/lib/python3.7/site-packages/torch/nn/modules/module.py", line 1051, in _call_impl
    return forward_call(*input, **kwargs)
  File "/data/run01/scz3687/openmmlab0171/mmdetection3d/plugin/futr3d/models/utils/transformer.py", line 157, in forward
    **kwargs)
  File "/HOME/scz3687/.conda/envs/open-mmlab0171/lib/python3.7/site-packages/torch/nn/modules/module.py", line 1051, in _call_impl
    return forward_call(*input, **kwargs)
  File "/data/run01/scz3687/openmmlab0171/mmdetection3d/plugin/futr3d/models/utils/transformer.py", line 215, in forward
    **kwargs)
  File "/HOME/scz3687/.conda/envs/open-mmlab0171/lib/python3.7/site-packages/torch/nn/modules/module.py", line 1051, in _call_impl
    return forward_call(*input, **kwargs)
  File "/HOME/scz3687/.conda/envs/open-mmlab0171/lib/python3.7/site-packages/mmcv/cnn/bricks/transformer.py", line 508, in forward
    **kwargs)
  File "/HOME/scz3687/.conda/envs/open-mmlab0171/lib/python3.7/site-packages/torch/nn/modules/module.py", line 1051, in _call_impl
    return forward_call(*input, **kwargs)
  File "/data/run01/scz3687/openmmlab0171/mmdetection3d/plugin/futr3d/models/utils/attention.py", line 241, in forward
    img_feats, reference_points, self.pc_range, kwargs['img_metas'])
  File "/data/run01/scz3687/openmmlab0171/mmdetection3d/plugin/futr3d/models/utils/attention.py", line 344, in feature_sampling
    lidar2img = lidar2img.view(B, num_cam, 1, 4, 4).repeat(1, 1, num_query, 1, 1)
RuntimeError: shape '[1, 4, 1, 4, 4]' is invalid for input of size 16
```
This is th log file.
[20220929_175225.log](https://github.com/Tsinghua-MARS-Lab/futr3d/files/9673998/20220929_175225.log)

xyaochen, 2022-09-29:
Could find out the size of lidar2img? What number of cameras did you use? It seems like the num_cam is equal to 1 in your datasets

shb9793, 2022-09-29:
Yes. My dataset only contains one camera. So I changed `num_cams` from 6 to 1.

Sorry. But I don't know how to get the size of 'lidar2img'. Because `img_metas` is not clear. Could you please tell me where should I find out the information of `img_metas` for our dataset.

```ruby
def feature_sampling(mlvl_feats, reference_points, pc_range, img_metas):
    lidar2img = []
    for img_meta in img_metas:
        lidar2img.append(img_meta['lidar2img'])
```

closed 2023-05-29 by xyaochen