Tsinghua-MARS-Lab/futr3d #21: Compatibility problem with mmdetection3d==0.13.0

APassbyDreg, 2022-11-01:
When working with mmdet3d==0.13.0, I got following error:

```
Traceback (most recent call last):
  File "tools/train.py", line 18, in <module>
    from mmdet3d.apis import train_model
ImportError: cannot import name 'train_model' from 'mmdet3d.apis' (/home/hzq/projects/futr3d/mmdet3d/apis/__init__.py)
Traceback (most recent call last):
  File "tools/train.py", line 18, in <module>
    from mmdet3d.apis import train_model
ImportError: cannot import name 'train_model' from 'mmdet3d.apis' (/home/hzq/projects/futr3d/mmdet3d/apis/__init__.py)
Traceback (most recent call last):
  File "tools/train.py", line 18, in <module>
    from mmdet3d.apis import train_model
ImportError: cannot import name 'train_model' from 'mmdet3d.apis' (/home/hzq/projects/futr3d/mmdet3d/apis/__init__.py)
Traceback (most recent call last):
  File "tools/train.py", line 18, in <module>
    from mmdet3d.apis import train_model
ImportError: cannot import name 'train_model' from 'mmdet3d.apis' (/home/hzq/projects/futr3d/mmdet3d/apis/__init__.py)
```

Indicating that mmdetection3d==0.13.0 is not the version to develop with. I found that `train_model` is added to mmdet3d.apis in version 0.14.0. Maybe readme.md should be updated to a correct requirement version.

cgl-cell, 2022-11-05:
which version are you using , can you achieve the results of the paper?

carry-all-coder, 2022-12-01:
save problem, ImportError: cannot import name 'train_model' from 'mmdet3d.apis' (/home/hzq/projects/futr3d/mmdet3d/apis/__init__.py) with mmdetection3d==0.13.0

closed 2023-05-29 by xyaochen