Tsinghua-MARS-Lab/futr3d #22: TypeError: NuScenesDatasetRadar: Unsupported format: None

deepmeng, 2023-01-07:
Hi, @xyaochen, thanks for your great work. I met an error as follows.

`Traceback (most recent call last):
  File "/home/dm/workspace/py36env4mmlab/lib/python3.6/site-packages/mmcv/utils/registry.py", line 69, in build_from_cfg
    return obj_cls(**args)
  File "/data/mm/project/futr3d/plugin/futr3d/datasets/nuscenes_radar.py", line 125, in __init__
    test_mode=test_mode)
  File "/data/mm/project/futr3d/z-dp/mmdetection3d/mmdet3d/datasets/custom_3d.py", line 89, in __init__
    self.data_infos = self.load_annotations(open(local_path, 'rb'))
  File "/data/mm/project/futr3d/plugin/futr3d/datasets/nuscenes_radar.py", line 174, in load_annotations
    data          = mmcv.load(ann_file)
  File "/home/dm/workspace/py36env4mmlab/lib/python3.6/site-packages/mmcv/fileio/io.py", line 57, in load
    raise TypeError(f'Unsupported format: {file_format}')
TypeError: Unsupported format: None

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "tools/train.py", line 244, in <module>
    main()
  File "tools/train.py", line 206, in main
    datasets = [build_dataset(cfg.data.train)]
  File "/data/mm/project/futr3d/z-dp/mmdetection3d/mmdet3d/datasets/builder.py", line 53, in build_dataset
    dataset = build_from_cfg(cfg, MMDET_DATASETS, default_args)
  File "/home/dm/workspace/py36env4mmlab/lib/python3.6/site-packages/mmcv/utils/registry.py", line 72, in build_from_cfg
    raise type(e)(f'{obj_cls.__name__}: {e}')
TypeError: NuScenesDatasetRadar: Unsupported format: None
`

It seems that the 'NuScenesDatasetRadar' registered failed or the NuScenes dataset was not processed properly.  I followed the rules
of mmdet3d and futr3d to process the NuScenes dataset.  Other models of 'cam_only', 'lidar_cam', and 'lidar_only' work fine.

Could you give me some advice?

deepmeng, 2023-01-07:
My runtime environment:

```
TorchVision: 0.9.0+cu111
OpenCV: 4.6.0
MMCV: 1.7.0
MMCV Compiler: GCC 7.3
MMCV CUDA Compiler: 11.1
MMDetection: 2.26.0
MMSegmentation: 0.29.1
MMDetection3D: 1.0.0rc6+4584dbc
spconv2.0: False

```

deepmeng, 2023-01-08:
solved already

closed 2023-01-08 by deepmeng

HaojiLan, 2024-02-17:
I have met the same problem, could you share your method with me?

wst20000804, 2024-04-01:
> solved already

怎么解决的？

azxcdewq123, 2024-06-23:
> solved already

请问这个问题是怎么解决的呢？我们也遇到相似的报错了