Tsinghua-MARS-Lab/futr3d #25: Corrupted lidar_cam.pth checkpoint file?

mrvangeerenstein, 2023-02-21:
Hi there,

I am trying to reproduce your results on the nuscenes validation set, and I downloaded the lidar_cam.pth checkpoint that is linked to in your readme. 
However, when loading this file using `torch.load()`, I get the error that: 

`RuntimeError: PytorchStreamReader failed reading zip archive: failed finding central directory`

Suprisingly, when I attempt to load the lidar_only checkpoint, it loads without any issues. 
When looking online for the reason of this error, I find that it is often caused by a corrupted/incorrectly-saved file. 
Could it be that the lidar_cam.pth checkpoint that you link to for downloading, is corrupted?
Are others able to load this checkpoint?
I tested this with python=3.8 and torch=1.10.0, and later with python=3.8 and torch=1.13.1.

Any suggestions?

xyaochen, 2023-02-25:
Sorry, it may encounter some error in uploading files previously. I have updated lidar_cam.pth. You can find its link in the original position.

mrvangeerenstein, 2023-02-28:
> Sorry, it may encounter some error in uploading files previously. I have updated lidar_cam.pth. You can find its link in the original position.

That's great. Thanks! I'll test it later today

closed 2023-04-05 by mrvangeerenstein