Tsinghua-MARS-Lab/futr3d #33: mmcv version

LYX-SOUL, 2023-06-21:
The package I installed is shown below：
[image]

**ERROR**: MMCV==2.0.0 is used but incompatible. Please install mmcv>=1.3.17, <=1.8.0.
How to solve this problem？Thanks

LYX-SOUL, 2023-06-21:
resloved

closed 2023-06-21 by LYX-SOUL

MinkyoungCho, 2023-06-25:
How did you resolve this issue? Can you let me know the versions of the modules (like your above screenshot)?

LYX-SOUL, 2023-06-29:
Available package versions：
[image]

sijieaaa, 2023-09-05:
mim install mmcv-full==1.6.2
mim install mmdet==2.28.0
mim install mmengine==0.7.4
mim install mmsegmentation==0.30.0