Tsinghua-MARS-Lab/futr3d #36: MMCV incompatibility issue

MinkyoungCho, 2023-06-26:
Hi! Thanks for your impressive work, this work is amazing!

While reproducing this work, I got the following error and saw several same issues have been raised up, but I could not find any clear solution in their posts. How can we solve the following issue?

`AssertionError: MMCV==1.6.0 is used but incompatible. Please install mmcv>=(1, 3, 13, 0, 0, 0), <=(1, 5, 0, 0, 0, 0).`
(FYI, if I simply upgrade the mmcv's version, the subsequent error msg shows I have to downgrade the version ...)

One more question: there is a DockerImage in this repo. Do you have plan for releasing docker image for this codebase? It would be really helpful to someone like me who is struggling with incompatibility issues...

MinkyoungCho, 2023-06-26:
resolved.

closed 2023-06-26 by MinkyoungCho

LIvanoff, 2024-05-26:
how?

> resolved.