Tsinghua-MARS-Lab/futr3d #46: mmcv version issue

Muhammadadeelaltaf, 2023-08-25:
When i am trying to create data through this command: python3 tools/create_data.py nuscenes --root-path ./mmdet3d/tools/configs/data/nuscenes --out-dir ./mmdet3d/tools/configs/data/nuscenes --extra-tag nuscenes
 it is giving me the following error: 
/home/citacav/futr3d/mmcv/mmcv/__init__.py:20: UserWarning: On January 1, 2023, MMCV will release v2.0.0, in which it will remove components related to the training process and add a data transformation module. In addition, it will rename the package names mmcv to mmcv-lite and mmcv-full to mmcv. See https://github.com/open-mmlab/mmcv/blob/master/docs/en/compatibility.md for more details.
  warnings.warn(
Traceback (most recent call last):
  File "tools/create_data.py", line 6, in <module>
    from tools.data_converter import kitti_converter as kitti
  File "/home/citacav/futr3d/tools/data_converter/kitti_converter.py", line 9, in <module>
    from mmdet3d.core.bbox import box_np_ops, points_cam2img
  File "/home/citacav/futr3d/mmdet3d/__init__.py", line 4, in <module>
    import mmdet
  File "/home/citacav/futr3d/mmdetection/mmdet/__init__.py", line 16, in <module>
    assert (mmcv_version >= digit_version(mmcv_minimum_version)
AssertionError: MMCV==1.7.0 is used but incompatible. Please install mmcv>=2.0.0rc4, <2.1.0.
If i install mmcv=2.0.0rc4 then it shows me the following error: 
File "/home/citacav/futr3d/mmdet3d/__init__.py", line 26, in <module>
    assert (mmcv_version >= digit_version(mmcv_minimum_version)
AssertionError: MMCV==2.0.0rc4 is used but incompatible. Please install mmcv>=1.5.2, <=1.7.0

arossbach10, 2024-01-18:
Any solution?

Adeel-farukh3, 2024-01-18:
@arossbach10  At that time, when I tried to implement it, I couldn't find any solution to these libraries regarding their version's compatibility.