Tsinghua-MARS-Lab/futr3d #51: some problem when evaluating Radar_cam model

kronaldo123, 2023-11-07:
I have flow the installation, everything is OK when I evaluating the lidar_cam model, the setting:
mmcv_full=1.6.0
mmsegmentation=0.30.0
mmdet=2.24.0
mmdet3d=1.0.0rc6  (follow the mmdet3d in project root)
cuda=11.1
pytorch=1.10

the result as shown below:
mAP: 0.5746
mATE: 0.4179
mASE: 0.4498
mAOE: 0.4368
mAVE: 0.4169
mAAE: 0.2883
NDS: 0.5863
Eval time: 1.8s

Per-class results:
Object Class        	AP    	ATE   	ASE   	AOE   	AVE   	AAE   
car                 	0.914 	0.171 	0.150 	0.076 	0.096 	0.064 
truck               	0.854 	0.178 	0.174 	0.020 	0.083 	0.000 
bus                 	0.992 	0.176 	0.091 	0.031 	0.582 	0.102 
trailer             	0.000 	1.000 	1.000 	1.000 	1.000 	1.000 
construction_vehicle	0.000 	1.000 	1.000 	1.000 	1.000 	1.000 
pedestrian          	0.882 	0.160 	0.254 	0.254 	0.169 	0.134 
motorcycle          	0.824 	0.228 	0.285 	0.416 	0.059 	0.005 
bicycle             	0.559 	0.196 	0.203 	0.134 	0.347 	0.000 
traffic_cone        	0.720 	0.069 	0.341 	nan   	nan   	nan   
barrier             	0.000 	1.000 	1.000 	1.000 	nan   	nan   

But I have got a problem when i was evaluating the Radar_cam model, the result I got can be shown below:
mAP: 0.3529
mATE: 0.6667
mASE: 0.7705
mAOE: 1.2119
mAVE: 0.4240
mAAE: 0.2929
NDS: 0.3610
Eval time: 2.2s

Per-class results:
Object Class        	AP    	ATE   	ASE   	AOE   	AVE   	AAE   
car                 	0.682 	0.390 	0.755 	1.530 	0.105 	0.079 
truck               	0.514 	0.456 	0.786 	1.413 	0.120 	0.043 
bus                 	0.492 	0.743 	0.866 	0.535 	0.517 	0.041 
trailer             	0.000 	1.000 	1.000 	1.000 	1.000 	1.000 
construction_vehicle	0.000 	1.000 	1.000 	1.000 	1.000 	1.000 
pedestrian          	0.558 	0.555 	0.323 	1.555 	0.332 	0.156 
motorcycle          	0.447 	0.527 	0.824 	1.449 	0.064 	0.011 
bicycle             	0.179 	0.641 	0.808 	1.425 	0.255 	0.013 
traffic_cone        	0.657 	0.354 	0.343 	nan   	nan   	nan   
barrier             	0.000 	1.000 	1.000 	1.000 	nan   	nan   
I found the mAOE is not the same as the paper , it is too large. I don't know what happen. I also met the same problem when I evaluating DETR3D model. I have found some of possible causing like:
1. mmdet3d have different parameter with version over than 1.0.0. 
2. Also maybe I did't set the config file in right way. Yes it is that I have do nothing with the config file.
Now I even think of the result of lidar_cam model is also have something wrong, I don't know what happen. Thanks

Zhh512, 2023-11-13:
[image]
i got the same problem ,have you solved the issue

kronaldo123, 2023-11-16:
> [image] i got the same problem ,have you solved the issue

Sorry, not yet

kronaldo123, 2023-12-19:
> ![image](https://private-user-images.githubusercontent.com/73172804/282407439-63c326c9-e87e-4d24-9716-d8ee85e137d3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

Hello, there is a problem  confuse me for a long time  that do u know the velocity estimation in their model, it seems like they don't have any temporal operation.

Zhh512, 2023-12-19:
> Hello, there is a problem confuse me for a long time that do u know the velocity estimation in their model, it seems like they don't have any temporal operation.

Radar point features contain velocity information, which does not require time difference to estimate .

kronaldo123, 2023-12-19:
> Radar point features contain velocity information, which does not require time difference to estimate .

Dear, Let's have look at  their code of Radar part.
as defined by config:
radar_use_dims = [0, 1, 2, 8, 9, 18]

related code is:
                # velocity compensated by the ego motion in sensor frame
                velo_comp = points_sweep[:, 8:10]
                velo_comp = np.concatenate(
                    (velo_comp, np.zeros((velo_comp.shape[0], 1))), 1)
                velo_comp = velo_comp @ sweep['sensor2lidar_rotation'].T
                velo_comp = velo_comp[:, :2]

                # velocity in sensor frame
                velo = points_sweep[:, 6:8]
                velo = np.concatenate(
                    (velo, np.zeros((velo.shape[0], 1))), 1)
                velo = velo @ sweep['sensor2lidar_rotation'].T
                velo = velo[:, :2]

                points_sweep[:, :3] = points_sweep[:, :3] @ sweep[
                    'sensor2lidar_rotation'].T
                points_sweep[:, :3] += sweep['sensor2lidar_translation']
                points_sweep_ = np.concatenate(
                    [points_sweep[:, :6], velo,
                     velo_comp, points_sweep[:, 10:],
                     time_diff], axis=1)
                points_sweep_list.append(points_sweep_)
        
        points = np.concatenate(points_sweep_list, axis=0)
        
        points = points[:, self.use_dim]

We can see that they just use velocity_compensate in model. 
Also, in their model named "DETR3D"， I have the same confusion. They can not use Radar in a camera only model, but they still predict velocity with a reg_branch(Linear layer) and also with no temporal related information evolved. 
I'm very happy someone could discuss this with me.

Zhh512, 2023-12-19:
> > > > ![image](https://private-user-images.githubusercontent.com/73172804/282407439-63c326c9-e87e-4d24-9716-d8ee85e137d3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI5NjczMjMsIm5iZiI6MTcwMjk2NzAyMywicGF0aCI6Ii83MzE3MjgwNC8yODI0MDc0MzktNjNjMzI2YzktZTg3ZS00ZDI0LTk3MTYtZDhlZTg1ZTEzN2QzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjE5VDA2MjM0M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQzOWJiMTc0YTExMTI5ZDA4MDA3MjY2MTQwNDkzNzlkMzVlMDJjNjU0MGIzYmNlMTY1NjExY2MxOTkyNzZkMTgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.w3uoKfvpGNyTbZvNbrBIeQo8N_JiWqN-L_OXa6SWae0) i got the same problem ,have you solved the issue
> > > 
> > > 
> > > Hello, there is a problem confuse me for a long time that do u know the velocity estimation in their model, it seems like they don't have any temporal operation.
> > 
> > 
> > Radar point features contain velocity information, which does not require time difference to estimate .
> 
> Dear, Let's have look at their code of Radar part. as defined by config: radar_use_dims = [0, 1, 2, 8, 9, 18]
> 
> related code is:
> 
> ```
>             # velocity compensated by the ego motion in sensor frame
>             velo_comp = points_sweep[:, 8:10]
>             velo_comp = np.concatenate(
>                 (velo_comp, np.zeros((velo_comp.shape[0], 1))), 1)
>             velo_comp = velo_comp @ sweep['sensor2lidar_rotation'].T
>             velo_comp = velo_comp[:, :2]
> 
>             # velocity in sensor frame
>             velo = points_sweep[:, 6:8]
>             velo = np.concatenate(
>                 (velo, np.zeros((velo.shape[0], 1))), 1)
>             velo = velo @ sweep['sensor2lidar_rotation'].T
>             velo = velo[:, :2]
>             points_sweep[:, :3] = points_sweep[:, :3] @ sweep[
>                 'sensor2lidar_rotation'].T
>             points_sweep[:, :3] += sweep['sensor2lidar_translation']
>             points_sweep_ = np.concatenate(
>                 [points_sweep[:, :6], velo,
>                  velo_comp, points_sweep[:, 10:],
>                  time_diff], axis=1)
> 
>             points_sweep_list.append(points_sweep_)
>     
>     points = np.concatenate(points_sweep_list, axis=0)
>     
>     points = points[:, self.use_dim]
> ```
> 
> We can see that they just use velocity_compensate in model. Also, in their model named "DETR3D"， I have the same confusion. They can not use Radar in a camera only model, but they still predict velocity with a reg_branch(Linear layer) and also with no temporal related information involved. I'm very happy someone could discuss this with me.
我觉得单帧图像确实是没法估计速度的，这也是他误差大的原因吧，然后就是补偿速度指的是被自车速度补偿过之后的速度，指的是目标的真实速度。你在完整数据集上测试这个模型了吗?如果测试了，希望你能分享一下结果
I don't think it's really possible to estimate the speed in a single frame image, which is why he has a large error I guess, and then there's the compensated speed which refers to the speed after being compensated by the autoroute speed, which refers to the real speed of the target. Have you tested this model on the full dataset? If you tested it, I hope you can share the results

kronaldo123, 2023-12-20:
> ![image](https://private-user-images.githubusercontent.com/73172804/282407439-63c326c9-e87e-4d24-9716-d8ee85e137d3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

感谢。不过我没测试这个模型，觉得有些不合理。我有问过官方但是他们还没有给到我回复。我正在考虑寻找其他的方案。